ghi-electronics/TinyCLR-Libraries #479: Touch move out of a button does not release the button hint

martincalsyn, 2020-03-24:
If you implement touch (as shown below) 

```
            var i2cDevice = I2cController.GetDefault().GetDevice(FT5xx6Controller.GetConnectionSettings());
            var gpioDevice = GpioController.GetDefault().OpenPin(SC20260.GpioPin.PJ14);
            var touch = new FT5xx6Controller(i2cDevice, gpioDevice);
            touch.TouchDown += this.Touch_TouchDown;
            touch.TouchMove += this.Touch_TouchMove;
            touch.TouchUp += this.Touch_TouchUp;

```

And connect it to UI as follows:

        private void Touch_TouchUp(FT5xx6Controller sender, GHIElectronics.TinyCLR.Drivers.FocalTech.FT5xx6.TouchEventArgs e)
        {
            if (this.touchState)
            {
                this.app.InputProvider.RaiseTouch(e.X, e.Y, TouchMessages.Up, DateTime.UtcNow);
                this.touchState = false;
            }
        }

        private void Touch_TouchMove(FT5xx6Controller sender, GHIElectronics.TinyCLR.Drivers.FocalTech.FT5xx6.TouchEventArgs e)
        {
            this.app.InputProvider.RaiseTouch(e.X, e.Y, TouchMessages.Move, DateTime.UtcNow);
        }

        private void Touch_TouchDown(FT5xx6Controller sender, GHIElectronics.TinyCLR.Drivers.FocalTech.FT5xx6.TouchEventArgs e)
        {
            if (!this.touchState)
            {
                this.app.InputProvider.RaiseTouch(e.X, e.Y, TouchMessages.Down, DateTime.UtcNow);
                this.touchState = true;
            }
        }

Then create a standard button:

            var txt = new Text(this.Frame.HeadingFont, label)
            {
                VerticalAlignment = VerticalAlignment.Center,
                HorizontalAlignment = HorizontalAlignment.Center,
            };

            var button = new Button()
            {
                Child = txt,
                Width = 100,
                Height = 50,
            };

Then when you touch down within the button, it will change to show the "pressed" visual state.  If you touch up (release) while still within the button, it will return to the "released" state.  However, if you touch down, and then drag your finger out of the button, the button will still show as "pressed" giving the user a confusing visual indication.

davemclaughlin, 2020-03-24:
That is the same as you get with standard windows. The button stays down and when you release, even when you move off the button, the click is triggered. We ran across this recently with a touch screen control system. The client spotted that the button stays down as he moved his finger off it.

martincalsyn, 2020-03-24:
The trouble is, the button is staying pressed even after you remove your finger from the screen.  On windows, drag out and release still triggers the click.  On TinyCLR, the button stays down even after you lift your finger. Either ignoring touch-up outside the button and restoring the unclicked state, or committing and restoring the state would be ok, but leaving the button looking pressed when no finger is on the screen seems like a bug.

davemclaughlin, 2020-03-24:
Ah ok. That is different and not correct so would be a bug.

mifmasterz, 2020-06-02:
little fix, if I use the same behaviour like in winform button it's like this https://youtu.be/XgEK3EJ8Tpc please take a look...

Palomino34, 2020-06-09:
Fixed by #607

commit ad5af1e0 (2020-06-09): Merge pull request #608 from ghi-electronics/#479_button_move_dont_release

Fixed #479.
commit c455054c (2020-06-09): Fixed #479.
closed 2020-06-09 by Palomino34