ghi-electronics/TinyCLR-Libraries #481: Can't build the libraries project

martincalsyn, 2020-03-24:
The projects each contain a packages.config file that references assemblies that are not available outside of GHI:

```
<package id="GHIElectronics.Tools.AssemblySigner" version="1.0.3" targetFramework="net452" developmentDependency="true" />
<package id="GHIElectronics.Tools.NuGetPacker" version="1.0.3" targetFramework="net452" developmentDependency="true" />
```

I'd love to contribute more PRs, but that's made harder because I can't build the debug versions or the nuget packages for testing (though I can build raw dlls by editing all of the package.config files) and just not pushing those changes.

SUGGESTION: Check in the nuget packager, and check in a signer that fails (succeeds?) silently if GHI signing keys are not found.

Palomino34, 2020-03-24:
We will take a look. Thanks.

martincalsyn, 2020-03-28:
This has been partially resolved with the changes Dat made in commit fa67658587d2861a5716ccd5821e14ed531314e3

closed 2020-05-13 by gus-ghielec