ghi-electronics/TinyCLR-Libraries #500: Project->Add->Class does not select the correct template

martincalsyn, 2020-03-30:
In the TinyCLR vsix (2.0 preview 4), when you right click on a project or project folder and select "Add" and then select "Class", you should get a template selection dialog that defaults to adding a new C# class.  Instead, it is defaulting to adding a new Xaml class:

[image]

The user then has to click on TinyCLR, which defaults to a BrainPad class; the user then has to select 'Class'. The 90% case is adding a plain old C# class, so that should be the default as it is for other projects.

martincalsyn, 2020-03-30:
Hmmm. This is not technically a 'libraries' problem. Is there another issues db for vsix problems?

Palomino34, 2020-06-09:
Do you know what field to control this issue in vstemplate? I tried some they don't work.

Thanks