ghi-electronics/TinyCLR-Libraries #503: Network stack requires delays in order to work properly

martincalsyn, 2020-04-02:
I was trying to retrieve SNTP time values, and had to use the code below (with delays).  If you remove the delays, this fails silently and ntpData is unmodified. With the delay statements in there, it works correctly.

```
            Array.Clear(ntpData, 0, ntpData.Length);
            ntpData[0] = 0x1B; //LeapIndicator = 0 (no warning), VersionNum = 3 (IPv4 only),
                               //    Mode = 3 (Client Mode)

            var addresses = Dns.GetHostEntry(ntpServer).AddressList;
            var ipEndPoint = new IPEndPoint(addresses[0], 123);
            var socket = new Socket(
                AddressFamily.InterNetwork,
                SocketType.Dgram,
                ProtocolType.Udp);

            socket.ReceiveTimeout = 3000;

            socket.Connect(ipEndPoint);

            Thread.Sleep(100); //Added to support TinyCLR OS.

            socket.Send(ntpData);

            Thread.Sleep(100); //Added to support TinyCLR OS.

            socket.Receive(ntpData);

            Thread.Sleep(100); //Added to support TinyCLR OS.

            socket.Close();
```

Palomino34, 2020-05-29:
We believe it is fixed from Preview6 which implemented timeout.

Can you please verify?

Palomino34, 2020-06-09:
Becuase timeout is implemented from Preview6, so couldn't reproduce any more.

Test by simple code below:
Server:
```
static void TestServerSocket()
        {
            Socket socket = new Socket(AddressFamily.InterNetwork, SocketType.Stream, ProtocolType.Tcp);            
            socket.Bind(new IPEndPoint(IPAddress.Any, 123));
            socket.Listen(4);

            int cnt = 0;

            while (true)
            {
                var clientSocket = socket.Accept();

                var dataRead = new byte[1024];

                var len = clientSocket.Receive(dataRead);

                Console.WriteLine(System.Text.UTF8Encoding.UTF8.GetString(dataRead, 0, len));

                clientSocket.Send(System.Text.UTF8Encoding.UTF8.GetBytes("this is for test " + cnt));

                cnt++;

            }
        }

```
Client:
```
static void TestClientSocket()
        {

            var dataRec = new byte[1024];

            var ipEndPoint = new IPEndPoint(IPAddress.Parse("192.168.1.87"), 123);
            var cnt = 0;

            while (true)
            {
                var socket = new Socket(
                    AddressFamily.InterNetwork,
                    SocketType.Stream,
                    ProtocolType.Tcp);

                var dataSend = UTF8Encoding.UTF8.GetBytes("I am SITCore " + cnt++);

                socket.ReceiveTimeout = 3000;

                socket.Connect(ipEndPoint);

                socket.Send(dataSend);

                var len = socket.Receive(dataRec);

                socket.Close();

                Debug.WriteLine("Rec: " + UTF8Encoding.UTF8.GetString(dataRec, 0, len));

            }
        }
```

closed 2020-06-09 by Palomino34