ghi-electronics/TinyCLR-Libraries #510: File information calls return "Not yet implemented"

martincalsyn, 2020-04-06:
I can't get file information (like last write time) because the underlying call returns a not-yet-implemented exception.  This is pretty core functionality.

```
        private bool UpdateFile(string url, FileInfo fi)
        {
            bool result = true;

            DateTime lastWriteTime = DateTime.MinValue;
            if (fi.Exists)
            {
                lastWriteTime = fi.LastWriteTimeUtc; <-- Throws an exception
            }

            this.CopyFromWebIfNewer(url, fi, lastWriteTime);

            return result;
        }

```

The exception originates here : 

Line 77 in FileSystemInfo.cs : 
`                this._nativeFileInfo = DriveInfo.GetForPath(this.m_fullPath).GetFileSystemEntry(this.m_fullPath);
`

Looks like it is not implemented in the native code component.

Palomino34, 2020-04-13:
fixed

closed 2020-04-13 by Palomino34

martincalsyn, 2020-04-13:
👏👍