ghi-electronics/TinyCLR-Libraries #523: Dropdown doesn't respond to direct touches

martincalsyn, 2020-04-11:
REPRO
Set up a touchscreen project
Create a dropdown with at least three items in it
Expand the dropdown
Try to touch directly on one of the items

EXPECTED
The touched item should become the new selected item

OBSERVED
The touched item is not selected
However, if you touch any other item, and then 'slide' your finger to the desired item, and then lift your finger, then the item where you lifted your finger will be selected
Also, the touch-up must occur on the text of the desired item.  In a normal UWP dropdown or listbox, touching anywhere on that line will select the item.

It looks like touch messages aren't being handled correctly for dropdowns.

mifmasterz, 2020-05-22:
hmm, I need to understand how it works first,, but the glide version of dropdown has been running in wpf. https://www.youtube.com/watch?v=8gby1FAtOMU but the touch move function is not functioning correctly so we can't choose the item that exceeds the screen height, need to change the algorthm.

mifmasterz, 2020-06-02:
now we can handle more items with paging in combobox.. https://youtu.be/SKlIE7UWguU 
I have tried the wpf dropdown, it's working but don't move your finger too fast or the item will not be selected..

Palomino34, 2020-07-27:
WNF. Touch to item directly will work. Touching on same row or column with item but not directly item, ListBoxItem won't trigger event which is the main to control item changed.

closed 2020-07-27 by Palomino34