ghi-electronics/TinyCLR-Libraries #525: Network operations on any thread block all threads

martincalsyn, 2020-04-11:
I've noticed that a consequence of the new TinyCLR api is that socket connect operations (and web requests, etc.) seem to block all threads.  That means that your app can end up completely unresponsive for 30 seconds or more if a connect call is made on any thread to an endpoint that is not currently reachable.

Previously, with the polling patterns in netmf, the API was ugly, but the fact that you were polling in .Net code for a status that existed in hardware or in native code meant that other .net threads were still running. Your app could still be responsive to user interactions, even if the network calls on other threads where stalled out.

Now, what I am seeing is that the app stops executing all .Net code while waiting on Connect in the network layer.  I think I understand why this is happening - it seems that the actual connect call is happening in native code, so the interpreter layer isn't running anymore for the duration of time that we are in native code. 

This seems pretty serious because it can appear that your app is hung. The whole point of the dispatcher model and background calls is so that your UX never becomes unresponsive.

To fix this, all potentially long-running native-code operations ought to be polled from .net code for status. Either that, or the native code needs to pump the IL interpreter periodically.

Palomino34, 2020-04-11:
Do you have any small example that we can try?

martincalsyn, 2020-04-11:
The test program is attached. Problem verified on an SC20260D

PH11 should blink at 10 Hz
PH6 should blink at 1Hz
The blinking occurs on a completely separate thread from the network operations.
The leds will blink for 10 seconds before starting the network.

You should point the web url at a machine that exists but which has no web server running.  That is, point it at a windows machine that isn’t running w3svc for instance. The goal is to get the network call (e.g., connect() call) to be slow.

When the connect occurs, the blinking will stop for 20-40 seconds, which means that the blink thread has been stopped while network calls are being made.

[NetworkTroubles.zip](https://github.com/ghi-electronics/TinyCLR-Libraries/files/4465619/NetworkTroubles.zip)

Palomino34, 2020-06-09:
We fixed socket.connect().

Other will be next release.

pdcompto1, 2021-01-26:
This issue still remains for UartController.  Any actions involving UART + Network are causing blocking issues.

TGlev, 2021-02-16:
Any update on this issue? Waiting for a fix...

gus-ghielec, 2021-02-16:
This is a large change that can introduce all kinds of new issues. We hope to get to it later on the year

Palomino34, 2021-03-12:
Only Connect/Send/Receive/Poll socket are non blocking.

closed 2021-03-12 by Palomino34