ghi-electronics/TinyCLR-Libraries #526: OnScreenKeyboard should have a 'cancel' button

martincalsyn, 2020-04-11:
I almost feel guilty for entering another OSK bug ... almost.

The on-screen-keyboard has a 'done' button, but no 'cancel' button.  If you mess up data entry and can't remember what the original value was, there's no way out for you.

There should be an 'Esc' or 'Cancel' button to allow users to bail out of data entry without making a change.

mifmasterz, 2020-05-19:
[image]
now, it has cancel button.. but still in https://github.com/Gravicode/GlideX

gus-ghielec, 2020-05-19:
Great! Are you going to be sending a PR?

mifmasterz, 2020-05-22:
yes, but too many changes I've been done in UI library.. so, I need to separate the changes first

Palomino34, 2020-06-09:
No need cancel button. Done button is enough. Closed.

closed 2020-06-09 by Palomino34