ghi-electronics/TinyCLR-Libraries #540: JSON cannot deserialize long integers

martincalsyn, 2020-05-04:
Reported here : https://forums.ghielectronics.com/t/json-deserialized-example/22630/24

Looks like a failed typecast trying to turn a read integer into a long in the target object.

#### Exception System.Exception - CLR_E_WRONG_TYPE (1) ####
#### Message: 
#### System.Reflection.FieldInfo::SetValue [IP: 0000] ####
#### GHIElectronics.TinyCLR.Data.Json.JsonConverter::PopulateObject [IP: 014f] ####
#### GHIElectronics.TinyCLR.Data.Json.JsonConverter::DeserializeObject [IP: 000c] ####
#### SC20260_AWS_Test_01.Program::Main [IP: 008f] ####
Ausnahme ausgelöst: "System.Exception" in mscorlib.dll
Ein Ausnahmefehler des Typs "System.Exception" ist in mscorlib.dll aufgetreten.

gus-ghielec, 2020-05-13:
@martincalsyn is this something you can help with or we need to look into it?

Palomino34, 2020-10-26:
Closed. Thank to @martincalsyn

closed 2020-10-26 by Palomino34