ghi-electronics/TinyCLR-Libraries #554: I2C : Timeout exception when using address 0x00

MikroBusNet, 2020-05-07:
Hello,

The MikroElektronika CapSense Click module is using Cypress CY8C201A0 chip, which has a default address of 0x00.
Using this address with TinyCLR 2.0 gives a timeout exception :

> Le thread '<Aucun nom>' (0x2) s'est arrêté avec le code 0 (0x0).
>     #### Exception System.Exception - CLR_E_TIMEOUT (1) ####
>     #### Message: 
>     #### GHIElectronics.TinyCLR.Devices.I2c.Provider.I2cControllerApiWrapper::WriteRead [IP: 0000] ####
>     #### GHIElectronics.TinyCLR.Devices.I2c.I2cDevice::WriteRead [IP: 0021] ####
>     #### GHIElectronics.TinyCLR.Devices.I2c.I2cDevice::Write [IP: 000c] ####
>     #### MBN.Modules.CapSenseClick::SetRegister [IP: 001e] ####
>     #### MBN.Modules.CapSenseClick::Init [IP: 0040] ####
>     #### Examples.Program::Main [IP: 0008] ####
> Exception levée : 'System.Exception' dans GHIElectronics.TinyCLR.Devices.I2c.dll

0x00 is indeed a special address but it should still be useable, to me.

Btw, the module address can/could be changed but it still needs to be accessed at address 0x00 the first time, so I'm stuck.

gus-ghielec, 2020-05-07:
Address 0x00 is a special address and should be treated as such. We need to study this to understand the 0x00 usage before making any changes.

MikroBusNet, 2020-05-09:
Address 0x00 is a special address and it seems I'm the only one using it. So don't bother with that, there are more serious features needing your attention.

gus-ghielec, 2020-05-09:
We have already purchased the same module you have and we will be working on it.

MikroBusNet, 2020-05-19:
Solved by @dat_tran : https://forums.ghielectronics.com/t/problem-with-i2c-address-0x00/22945/21?u=bec_a_fuel

You can close this issue. With apologies. 😞

closed 2020-05-19 by gus-ghielec