ghi-electronics/TinyCLR-Libraries #571: Tiny File system

gus-ghielec, 2020-05-15:
The community contributed tiny file system to be used with smaller flash and eeprom. Perhaps we can bring back as a nuget.

MikroBusNet, 2020-05-18:
Gus,
FYI, we have a working TinyFileSystem driver working. It will come online soon with our other Click drivers.

Here is a code example, using a Flash3 Click on SC20260D :

```
class Program
    {
        private static TinyFileSystem _tfs;

        public static void Main()
        {
            _tfs = new TinyFileSystem(new FlashMemory(Hardware.SC20260_1));
            if (_tfs.CheckIfFormatted())
            {
                Debug.WriteLine("Filesystem OK. Mounting...");
                _tfs.Mount();
                Debug.WriteLine("Mounted. Now reading settings.dat file...");
                if (!_tfs.Exists("settings.dat"))
                    return;
                using (Stream fs = _tfs.Open("settings.dat", FileMode.Open))
                using (var rdr = new StreamReader(fs))
                {
                    System.String line;
                    while ((line = rdr.ReadLine()) != null)
                    {
                        Debug.WriteLine(line);
                    }
                }
                TinyFileSystem.DeviceStats tfsStats = _tfs.GetStats();
                Debug.WriteLine("Stats :");
                Debug.WriteLine($"\tBytes free : {tfsStats.BytesFree}");
                Debug.WriteLine($"\tBytes orphaned : {tfsStats.BytesOrphaned}");
            }
            else
            {
                Debug.WriteLine("Formatting");
                _tfs.Format();
                Debug.WriteLine("Creating file");
                using (Stream fs = _tfs.Create("settings.dat"))
                {
                    using (var wr = new StreamWriter(fs))
                    {
                        wr.WriteLine("<settings>");
                        wr.WriteLine("InitialPosX=200");
                        wr.WriteLine("InitialPosY=150");
                        wr.WriteLine("</settings>");
                        wr.Flush();
                        fs.Flush();
                    }
                }
                Debug.WriteLine("FileCreated");
            }

            Thread.Sleep(Timeout.Infinite);
        }
    }

```
And the output :

> Filesystem OK. Mounting...
> Mounted. Now reading settings.dat file...
> <_settings>
> InitialPosX=200
> InitialPosY=150
> <_/settings>
> Stats :
> 	Bytes free : 16775168
> 	Bytes orphaned : 1024

The constructor takes a "Storage" class which, in our drivers, includes :

- All Flash Click modules
- All Eeprom modules
- Any "onboard Flash"

gus-ghielec, 2020-05-18:
Sounds great but shouldn't TinyFileSystem be a separate NuGet that can be used with any flash, providing the user implements read/write sector?

MikroBusNet, 2020-05-18:
So far, we provide source files, not NuGet yet. But it's something we are thinking at.

And, as you say, it can be used with any flash or any eeprom (and any memory that preserves data upon reset/power-cycle), as long as you provide the read/write methods.

That's exactly what the FlashClick drivers does. And this driver is unique for all Flash chips or modules.
It can be used as a standalone memory chip or as a storage location for TinyFileSystem.

gus-ghielec, 2020-05-19:
What I mean here is that I think these should be two separate libraries. One is the file system driver, which is what this issue is about. Second is the click driver for the specific click modules. There are multiple click modules with flash chips and they all can utilize the tiny file system.

Is the file system something you implemented or it is the same one that was contributed on codeshare years ago?

MikroBusNet, 2020-05-19:
I think we don't understand each other :)

So yes, there are two separate things : TFS on one side and "memory drivers" on the other side.

The code sample above was a simple example showing the use of TFS with a Flash Click. I could have shown the same code with an Eeprom Click or ans SRam Click.
TFS needs "something" to read/write its informations, it cannot work "as is" or standalone.

Our TFS implementation is heavily based on the work from Chris Taylor (Tailorza). It has been adapted to TinyCLR 2.0. I would say that 95% of the code is from him. Perhaps even more.
He has credits in the files, btw.

gus-ghielec, 2020-05-19:
That is great. Looking forward to seeing all the drivers.

Palomino34, 2020-08-24:
Added

closed 2020-08-24 by Palomino34