ghi-electronics/TinyCLR-Libraries #575: GPS library

gus-ghielec, 2020-05-16:
Need a tightly built, efficient library, to parser GPS streams

mifmasterz, 2020-05-19:
ported from netduino sample to tinyclr : https://github.com/mifmasterz/GPSParser.git 
I haven't tried it with real gps module, please take it

gus-ghielec, 2020-05-19:
There has been multiple GPS parser over the year. The challenge here is writing an efficient one that takes under concentration system heap fragmentation. GPS is very simple but at if the stream comes in once per second, non stop.

I did not review your code so I do not know. I also see you have included regular expressions so this ties into #556

mifmasterz, 2020-05-22:
this parser is from netduino version, so it must be worked for netmf, but yes we need to improve it's code for TinyCLR

closed 2020-11-24 by gus-ghielec