ghi-electronics/TinyCLR-Libraries #577: Support AutoIP

gus-ghielec, 2020-05-18:
Determine the advantage of adding AutoIP vs using DHCP and the benefit outweighs the cost add support. This was requested here https://forums.ghielectronics.com/t/apipa-autoip/23003

gus-ghielec, 2020-05-18:
Reading this, there is nothing to do really! The user can set the address to what they desire! Investigate further.

https://manuals.konicaminolta.eu/ineo-4000P-4700P/EN/ntwk_guide/what-is-auto-ip-topic.html

poundy, 2020-05-19:
As discussed in the forum, there's a need prior to setting an IP address on the interface to do the ARP broadcast and monitor responses to confirm a chosen random IP in the zeroconf designation isn't in use on the broadcast zone. 

RFC 3927 page 11, section 2.2.1 in http://files.zeroconf.org/draft-ietf-zeroconf-ipv4-linklocal.txt 1 is the best reference I’ve seen, although almost word for word same as RFC 5227 https://www.ietf.org/rfc/rfc5227.txt

Palomino34, 2020-12-02:
Added.

closed 2020-12-02 by Palomino34