ghi-electronics/TinyCLR-Libraries #581: USB DeviceConnectionStatus.Disconnected event not fired

MikroBusNet, 2020-05-21:
On the SC20260D, if you insert an USB stick, you get the DeviceConnectionStatus.Connected event fired, as expected.
But if you remove the USB stick, then you won't get the Disconnected event.

Then, it seems that a watchdog timeout is coming in as the VS output eventually says :

> _"The debugger engine failed to receive any debug events from the debugging target"_.

Using USB debugging, it is around 50 seconds later, while using Serial debugging gives something like 15-20 seconds.

In both cases, the board needs a reset to be able to deploy again.

gus-ghielec, 2020-05-21:
Can you provide a very small sample program that only shows this issue? It is working fine for us.

MikroBusNet, 2020-05-21:
Here it is :

```
class Program
    {
        private static readonly AutoResetEvent _usbReady = new AutoResetEvent(false);
        private static IDriveProvider drive;
        private static UsbHostController usbHostController;
        private static DriveInfo driveInfo;
        private static StorageController storageController;

        static void Main()
        {
            TestUSBHost();

            Thread.Sleep(Timeout.Infinite);
        }

        private static void TestUSBHost()
        {
            Debug.WriteLine($"{DateTime.Now} : Waiting for USB");
            usbHostController = UsbHostController.GetDefault();
            usbHostController.OnConnectionChangedEvent += UsbHostController_OnConnectionChangedEvent;
            usbHostController.Enable();
            _usbReady.WaitOne();
            Debug.WriteLine($"{DateTime.Now} : USB ready\n");
        }

        private static void UsbHostController_OnConnectionChangedEvent(UsbHostController sender, DeviceConnectionEventArgs e)
        {
            switch (e.DeviceStatus)
            {
                case DeviceConnectionStatus.Connected:
                    switch (e.Type)
                    {
                        case BaseDevice.DeviceType.MassStorage:

                            storageController = StorageController.FromName(SC20260.StorageController.UsbHostMassStorage);
                            drive = FileSystem.Mount(storageController.Hdc);
                            driveInfo = new DriveInfo(drive.Name);

                            Debug.WriteLine($"Free: {driveInfo.TotalFreeSpace}");
                            Debug.WriteLine($"TotalSize: {driveInfo.TotalSize}");
                            Debug.WriteLine($"VolumeLabel: {driveInfo.VolumeLabel}");
                            Debug.WriteLine($"RootDirectory: {driveInfo.RootDirectory}");
                            Debug.WriteLine($"DriveFormat: {driveInfo.DriveFormat}\n");

                            _usbReady.Set();

                            break;

                        default: break;
                    }
                    break;

                case DeviceConnectionStatus.Disconnected:
                    Debug.WriteLine("Device Disconnected");
                    //Unmount filesystem if it was mounted.
                    //FileSystem.Unmount(storageController.Hdc);
                    break;

                case DeviceConnectionStatus.Bad:
                    Debug.WriteLine("Bad Device");
                    break;
            }
        }
}

```

gus-ghielec, 2020-05-21:
Have you tried more than one drive? Do they all fail? What type of drive is it? The ones we tested work fine.

MikroBusNet, 2020-05-21:
I've tried with https://www.amazon.fr/basicXL-BXL-CARDRW2B-Lecteur-carte-pour/dp/B004DUNMDU and https://www.amazon.fr/Corsair-Cl%C3%A9-USB-Voyager-CMFVYGT3A-128GB/dp/B008MCZIEQ

Both detected and working fine with dev board. At least for basic r/w operations.

gus-ghielec, 2020-05-21:
One of them is available in USA. We will get one and test!

MikroBusNet, 2020-05-21:
Get the Corsair one 'if possible), it is a great device !!!

closed 2020-05-26 by gus-ghielec