ghi-electronics/TinyCLR-Libraries #644: In Pins PH11 pin is missing for SC20260D board

bauland, 2020-07-19:
In rc1, there is no SC20260.GpioPin.PH11 (which is connected on led) in Pins library.

Palomino34, 2020-07-19:
What do you mean "no SC20260.GpioPin.PH11"?

I have checked, it is there. 

You can check Demo on SC20260, I believe this project also uses  SC20260.GpioPin.PH11.

bauland, 2020-07-19:
Sorry, it is my fault: order of variables is incorrect:
[image]
but it exists indeed:
[image]
even it is not at expected place (order is not what I believe !). 
Issue can be closed and I owe you some beers !

Palomino34, 2020-07-20:
Not a bug.

closed 2020-07-20 by Palomino34