ghi-electronics/TinyCLR-Libraries #663: Feature request : extract "StorageDriver" from TinyFileSystem nuget

MikroBusNet, 2020-08-11:
Our TFS implementation (that you have included in your librairies, thank you !), the StorageDriver needed by this module is also used by all our memory chips drivers : Flash, Eeprom, SRam, etc...
But those don't need all the TFS stuff. They only need the StorageDriver abstract class.

So, if this StorageDriver class could be extracted from TFS it would be nice.

Palomino34, 2020-08-11:
Isn't StorageController class strong enough? 

I think this class can handle all your Flash, Eeprom, SRam, etc... by implementing IStorageControllerProvider

MikroBusNet, 2020-08-11:
I would not see it this way, in fact.

The TFS block driver **_does need_** the current StorageDriver implementation, so unless we recode it entirely, it should not change. Which means that the memory modules pretending to be used by TFS **_have to_** implement this interface.
So I would not change it to IStorageControllerProvider either.

Really, the solution is very simple : by taking StorageDriver out of TFS nuget, it can be used/implemented by many other drivers.
I would even say that since it's a generic interface (a class, in fact), it does not have to be in a particular implementation (TFS, here).
I would even say that by integrating the StorageDriver in the TFS nuget, you are more or less restricting its use to TFS only.

Do you see what I mean ?

To be honest, I won't insist that much on this anyway. If you don't want to extract the StorageDriver, I will simply not use your nuget but rather our TFS implementation. 
Please do not take it the wrong way : this is not a criticism or a "threat". You may have good reasons to not change anything. This would not satisfy me but since I have another solution 99.99% compatible, I would not really care.

Palomino34, 2020-09-23:
StorageDriver is removed.

commit f979b9a3 (2020-09-23): Fixed #663. Remove StorageDriver
closed 2020-09-23 by Palomino34