ghi-electronics/TinyCLR-Libraries #674: MQTT should pass the topic to the received message function

davemclaughlin, 2020-09-08:
As MQTT allows the use of wildcards, we need to get the topic as part of the received message to determine where the payload came from. The current implementation uses a packetId to indicate which topic it belongs to. This works fine for complete topsics but if we subscribe to something like roomsensors/# to get roomsensors/temperature and roomsensors/humidity, for example, the current code won't find a match for these when the messages arrive. Passing the topic to the receive handler will allow the decoding of each wildcard entry.

Palomino34, 2020-10-06:
Hello, to make sure we understand you, are you talking about subscribe a topic or receiving  a message?
- Subscribe a topic: I believe we support  wildcards as well.
- Receiving a message: From the message received, there is a field LastWillTopic will tell which topic it belong to.

We are improving MQTT so, if you see anything need to be corrected, just let us know.

Palomino34, 2020-10-06:
OK, we see, it is internal so can't access from different namespace. We will fix it. Thank.

Palomino34, 2020-10-06:
Hi, we found the issue and there are a lot of changed in the branch:
ghi-electronics/TinyCLR-Libraries/tree/mqtt_pass_topic_eventhandler

When you have time, can you please try to check that branch to see if everything is OK to you?

Thanks for your help!

Palomino34, 2020-10-09:
Added.

closed 2020-10-09 by Palomino34

qvmsoft, 2020-10-09:
Excellent. Everything is working. Thanks.