ghi-electronics/TinyCLR-Libraries #718: Ftp doesn't work with the host ftp://speedtest.tele2.net/

Palomino34, 2020-10-28:


Palomino34, 2021-04-01:
Need verify

Palomino34, 2021-04-07:
The code below should work but still raise exception:
```cs
static void DoTestFtpDownload()
        {
            // Download

            // Get the object used to communicate with the server.
            FtpWebRequest request = (FtpWebRequest)WebRequest.Create("ftp://speedtest4.tele2.net/512KB.zip");
            request.Method = WebRequestMethods.Ftp.DownloadFile;

            // This example assumes the FTP site uses anonymous logon.
            request.Credentials = new NetworkCredential("anonymous", "janeDoe@contoso.com");

            FtpWebResponse response = (FtpWebResponse)request.GetResponse();

            Stream responseStream = response.GetResponseStream();
            StreamReader reader = new StreamReader(responseStream);

            var output = reader.ReadToEnd();
            Debug.WriteLine(output);

            Debug.WriteLine("Download Complete");

            reader.Close();
            response.Close();
        }

        static void DoTestFtpUpload()
        {
            //Get the object used to communicate with the server.
            FtpWebRequest request = (FtpWebRequest)WebRequest.Create("ftp://speedtest4.tele2.net/upload/testfile.txt");
            request.Method = WebRequestMethods.Ftp.UploadFile;

            // This example assumes the FTP site uses anonymous logon.
            request.Credentials = new NetworkCredential("anonymous", "janeDoe@contoso.com");

            // Copy the contents of the file to the request stream.
            byte[] fileContents;
            using (StreamReader sourceStream = new StreamReader(@"E:\TinyCLR\Example\Desktop\DoTestFtp\testfile.txt"))
            {
                fileContents = Encoding.UTF8.GetBytes(sourceStream.ReadToEnd());
            }

            request.ContentLength = fileContents.Length;

            using (Stream requestStream = request.GetRequestStream())
            {
                requestStream.Write(fileContents, 0, fileContents.Length);
            }

            using (FtpWebResponse response = (FtpWebResponse)request.GetResponse())
            {
                Debug.WriteLine("Upload File Complete ");
            }
        }
```

AndreMarschalek, 2021-04-07:
which exception at which line?

btw this host is the reason why i have opened https://forums.ghielectronics.com/t/networking-socket-issue/23625 with the result of https://github.com/ghi-electronics/TinyCLR-Libraries/issues/925

Palomino34, 2021-04-07:
[image]

AndreMarschalek, 2021-04-07:
the reason for this exception is "ServerIP = IPAddress.Parse(uri.Host);   // throws ArgumentException" which does not work for dns names (because its clearly not an ip address)

```
                if (uri.HostNameType == UriHostNameType.Dns) {
                    System.Diagnostics.Debug.WriteLine("Status: Resolving address of " + uri.Host);
                    IPHostEntry iPHostEntry = Dns.GetHostEntry(uri.Host);    // throws InvalidOperationException
                    ServerIP = iPHostEntry.AddressList[0];
                }
                else if (uri.HostNameType == UriHostNameType.IPv4) {
                    ServerIP = IPAddress.Parse(uri.Host);   // throws ArgumentException
                }
                else {
                    throw new ArgumentException("HostNameType not supported.");
                }
```
i will do a new pr as the existing code does not reflect the latest changes

AndreMarschalek, 2021-04-07:
https://github.com/ghi-electronics/TinyCLR-Libraries/pull/985