ghi-electronics/TinyCLR-Libraries #719: Memory seems to be leaking quite fast

TGlev, 2020-10-29:
Hi,

In an attempt at figuring out a memory leak issue I have been having I made a very simple program to verify something:

```
using GHIElectronics.TinyCLR.Native;
using System.Diagnostics;
using System.Threading;

namespace MemTest
{
    class Program
    {

        static void Main()
        {
            while(true)
            {
                Debug.WriteLine($"Free: {Memory.ManagedMemory.FreeBytes}, used: {Memory.ManagedMemory.UsedBytes}");
                Thread.Sleep(100);
            }
        }
    }
}

```
When running this program, you can see in the output that something is RAPIDLY using free memory. This happens until the free memory reaches a low amount at which point the GC is triggered I guess.

[image]

Is this normal behaviour? If no, is this a bug or am I overlooking something?

TGlev, 2020-10-29:
I enabled GC messages and this is the output when it runs:

```
GC: 1msec 134832 bytes used, 372992 bytes available
Type 11 (STRING              ):    320 bytes
Type 13 (CLASS               ):   1216 bytes
Type 14 (VALUETYPE           ):    144 bytes
Type 15 (SZARRAY             ):    912 bytes
  Type 03 (U1                  ):    160 bytes
  Type 04 (CHAR                ):    352 bytes
  Type 07 (I4                  ):     48 bytes
  Type 13 (CLASS               ):    352 bytes
Type 17 (FREEBLOCK           ): 372992 bytes
Type 18 (CACHEDBLOCK         ):     96 bytes
Type 19 (ASSEMBLY            ):   7344 bytes
Type 1B (REFLECTION          ):     64 bytes
Type 1D (DELEGATE_HEAD       ):     48 bytes
Type 20 (BINARY_BLOB_HEAD    ): 122240 bytes
Type 21 (THREAD              ):    400 bytes
Type 22 (SUBTHREAD           ):     48 bytes
Type 23 (STACK_FRAME         ):   1264 bytes
Type 36 (APPDOMAIN_HEAD      ):     80 bytes
Type 38 (APPDOMAIN_ASSEMBLY  ):    656 bytes
```

gus-ghielec, 2020-10-29:
What you are seeing is very normal and that is how a managed system works. In your loop, you create strings (memory) and then they get discarded. Then the GC runs and cleans up all discarded memory. A memory leak means your memory goes down even if you run the GC.

closed 2020-10-29 by gus-ghielec

Palomino34, 2020-10-29:
As long as it goes back to full then it is normal, I think.

[7b6fc1e9f8848ec3b421ba6388bc0aa0]<https://user-images.githubusercontent.com/10108352/97545854-e2db2400-19cb-11eb-80f3-46de75920350.gif>