ghi-electronics/TinyCLR-Libraries #742: Improve UI

Palomino34, 2020-11-20:
- DataGrid
- Slider

davemclaughlin, 2020-12-07:
Toggle Button
Visual state on onscreen keyboard
Transparent background for images support

davemclaughlin, 2020-12-07:
Be able to change button graphic images for up, down and disabled states.
Tint option for buttons like the old Glide interface
Graphical loader similar to the old Glide with XML files. This is way faster than manually creating each button, text image, etc. 
GUI editor for the UI. Glide Studio from BlackdogSpark was excellent but sadly he never released the source code

gus-ghielec, 2020-12-12:
The lack of visual feedback on the buttons for the OnScreenKeyboard leaves the operator with a lack of feel if you can call it that.