ghi-electronics/TinyCLR-Libraries #756: Path.Combine does not de-duplicate separators

martincalsyn, 2020-12-05:
REPRO:
            var leftpart = @"test\path\";
            var rightpart = "index.html";
            var result = Path.Combine(leftpart, rightpart);
            Debug.WriteLine(result);

EXPECTED:
result == test\path\index.html

RESULT:
correct result on dotnet core and framework
result == test\path\\index.html on TinyCLR

TinyCLR should de-duplicate path separators.

gus-ghielec, 2020-12-05:
Both results are the same on your email but I am guessing they're is an extra / in TinyCLR

Regards,

Gus Issa
---
Please excuse brevity or typos, this was entered on a phone!

martincalsyn, 2020-12-05:
Interesting.  Markdown failure.  I typed it write but markdown messed with it.
Anyway, I edited the original report. The 'Result' section correctly shows the error now.

Palomino34, 2020-12-16:
As quick test, it look good to me.

[image]

closed 2020-12-16 by Palomino34