ghi-electronics/TinyCLR-Libraries #757: Calling Socket.Listen with an invalid value results in an unexpected default

martincalsyn, 2020-12-06:
If you call Socket.Listen() on the ATWINC15x0 with a value > 7 (the ATWIN's maximum), it looks like the listen queue gets set to 2. The third connection attempted in parallel hangs until one of the first two connections is closed.

Using a value >2 and <= 8 results in the expected ability to accept more TCP connections up to the specified limit.

If an unsupported value is passed to Socket.Listen() [a value greater than 7], it should throw an argument exception and not default to some other value.

Palomino34, 2020-12-23:
How do you reproduce the issue?

We have tested listening 3-4 socket, seems to us it work fine.

Also, we are using Ethernet bypass mode, so the listen socket can be up to 128 which independent to limitation on Winc1500 .

Palomino34, 2021-02-12:
Could not reproduce as my last comment.

closed 2021-02-12 by Palomino34