ghi-electronics/TinyCLR-Libraries #759: Stream writes to network sockets randomly throw

Palomino34, 2020-12-07:
```
public static void CopyTo(this Stream src, Stream dest)
        {
            int maxChunkSize = 0x100;
            int size = (src.CanSeek) ? Math.Min((int)(src.Length - src.Position), maxChunkSize) : maxChunkSize;
            byte[] buffer = new byte[size];
            int n;
            do
            {
                n = src.Read(buffer, 0, buffer.Length);
                if (n == -1) n=0;
                dest.Write(buffer, 0, n);
            } while (n != 0);
        }
```

If src is a stream on an SD card for a file that is largish (say, 200 K) and dest is a network socket stream.
Then
•	if maxChunkSize is small (0x100) then the stream gets copied successfully
•	if maxChunkSize is 0x2000 then the .Write() throws an IO exception (with no explanatory message),
•	if maxChunkSize is 0x1000, then it gets a bit further, but still throws.

martincalsyn, 2020-12-10:
Duplicate of Issue #765 (sorry - I didn't see this one before entering 765)
This issue is now resolved (PR #766)

Palomino34, 2020-12-11:
Done

closed 2020-12-11 by Palomino34