ghi-electronics/TinyCLR-Libraries #779: GC does not dispose unmanaged memory

MBNSoftware, 2020-12-13:
As said in the docs :

> **Graphical Memory**
> 
> When the [graphics](https://docs.ghielectronics.com/software/tinyclr/tutorials/graphics.html) engine detects available external memory, it automatically uses it. Also, garbage collection will dispose of unmanaged graphics buffers automatically. You do not have to dispose of unmanaged graphics buffers like you do for unmanaged non-graphic buffers.

As we can see below, 153 KB have been allocated :

>     +-----------+------------+------------+
>     | Memory    | Used       | Free       |
>     +-----------+------------+------------+
>     | Managed   |    498,496 |      9,296 |
>     | Unmanaged |    153,864 | 33,400,540 |
>     +-----------+------------+------------+
 
Now, after som time, GC has run and freed managed memory. But not unmanaged memory used by the screen.Buffer :
 
>     +-----------+------------+------------+
>     | Memory    | Used       | Free       |
>     +-----------+------------+------------+
>     | Managed   |    158,896 |    348,896 |
>     | Unmanaged |    153,864 | 33,400,540 |
>     +-----------+------------+------------+

I have tried to Dispose() the screen object, but this did not change anything.

Palomino34, 2020-12-16:
If the graphic created from Bitmap, you have to Dispose the Bitmap.

closed 2020-12-16 by Palomino34

Palomino34, 2020-12-16:
It is similar to Dispose SPI doesn't release Chipselect. 

So if Graphics is created from bitmap, then dispose graphics has nothing effect to bitmap.

MBNSoftware, 2020-12-16:
Ok. Understood.
Thank you !

Perhaps something in the doc to emphasize on this ?

Palomino34, 2020-12-16:
Not needed really, 

GC will clean the bitmap objects if not detect any references. 

```
var gfx = Graphics.FromImage(new Bitmap(240, 320));

gfx.Dispose();

GC.Collect();
GC.WaitForPendingFinalizers();
```

MBNSoftware, 2020-12-16:
Ok. But the doc does talk about the "graphic buffer", not the bitmap. And if you talk about buffer, then you immediately think at the `screen.Buffer` that you send to the device, not the highest level bitmap object.

Hence my request to at least clarify the terms. But it's not that important, no worry.