ghi-electronics/TinyCLR-Libraries #783: Additional Crypto functions needed for JWT

martincalsyn, 2020-12-15:
Additional crypto features need to be exposed in order to support generation of JWT tokens for use with popular web services, including Google IoT

SHA256 hash calculation
HMAC hash calculation
RSA signature calculation (at least RSA256)

Palomino34, 2020-12-15:
Does RSA "https://docs.ghielectronics.com/software/tinyclr/tutorials/cryptography.html" work?

martincalsyn, 2020-12-18:
Those functions work fine for encrypt/decrypt, but we need RSA256 signing (generating a signature hash, not an encrypted payload). Also need HMAC signing. There are also elliptic curve signing algorithms that are used with JWT, but if the goal is Google IoT, then RSA256 is sufficient to start with.