ghi-electronics/TinyCLR-Libraries #795: Mac address values are not consistent

martincalsyn, 2020-12-21:
This code will give two different answers:

```
                    var mac = Winc15x0Interface.GetMacAddress();

                    var ifprops = networkController.GetInterfaceProperties();
                    var mac = ifprops.MacAddress;

```

The first one will give you the MAC addr that is provided with the Winc15x0 chip, and the second will give all zeros until you complete the initialization.  It would seem that if a valid MAC addr is available, then the second call should also return that value.

martincalsyn, 2020-12-21:
And this returns yet another value after an AP client connects:

```
        private void WifiSettings_AccessPointClientConnectionChanged(NetworkController sender, IPAddress clientAddrres, string macAddress)
        {
            Debug.WriteLine("AP Client Connected : " + clientAddrres.ToString() + " mac: " + macAddress);

```

Palomino34, 2020-12-21:
NetworkController.MacAddress  only valid after completed initialize.  

```
private void WifiSettings_AccessPointClientConnectionChanged(NetworkController sender, IPAddress clientAddrres, string macAddress)
        {
            Debug.WriteLine("AP Client Connected : " + clientAddrres.ToString() + " mac: " + macAddress);
```

That MAC is from client connect to AP, not from WiFi.

martincalsyn, 2020-12-21:
Yup - just realized that - color me sheepish.
However, the original issue still stands.  The two APIs return different values (and all zeros isn't really useful, so I feel like that API should return the default from the winc chip.

gus-ghielec, 2021-03-24:
there isn't much to fix here. Maybe a note in docs that no one will likely see!

closed 2021-03-24 by gus-ghielec