ghi-electronics/TinyCLR-Libraries #807: Clear screen when initialized

gus-ghielec, 2021-01-04:
Internal graphics can show garbage on the screen when initialized. Clear the screen buffer

MBNSoftware, 2021-01-09:
It could be a optional parameter, set to true (clear screen) by default.
Sometimes it's useful to see garbage on the screen : it clearly shows where you are writing on the display.

Or, if you don't like to see garbage, add a Clear(RGBColor) overload ?

gus-ghielec, 2021-01-10:
In a commercial use having garbage is not okay. What you are asking for is good for debugging only. You can subbing flush something useful instead.

Palomino34, 2021-02-12:
- We fixed in Native display.
- Other displays (N18, ILI9341...), we won't fix because those screen don't use internal buffer. It takes the buffer from user directly, and transfer to the target devices. We don't want to create an internal for temporary buffer just for flushing to clear the screen.

closed 2021-02-12 by Palomino34