ghi-electronics/TinyCLR-Libraries #809: Track changing graphics

gus-ghielec, 2021-01-04:
The graphics library supports partial update through Flush(x, y, width, height). This great but how do we know what pixels have changed?

```
Graphics.WriteText(5, 10, "Hello");
Graphics.Flush(?,?,?,?);
```
The internal engine should track the lowest/highest changing pixels on x and y and then it can easily provide this info. Even better, automatically send the info so the user will only need to do Flush()

MBNSoftware, 2021-01-09:
Public read-only properties would be useful as well, as we may not flush immediately after a graphic action.

gus-ghielec, 2021-01-10:
This will only let track and not flush. It is up to you to flush when need

gus-ghielec, 2021-04-01:
The user have better tracking on what they need updated

closed 2021-04-01 by gus-ghielec