ghi-electronics/TinyCLR-Libraries #820: N18 display speed

gus-ghielec, 2021-01-09:
Devices such as FEZ Bit may need a faster update speeds and they are okay with 8BPP. We already sorry color space conversion. Give the user the option to select 8BPP and the color map table.

Even do 4BPP if the controller supports it. I think it does not

gus-ghielec, 2021-01-09:
Looks like 8BPP is but supported but 12BPP is. Determine if the additional memory usage outweighs the speed improvement. 128×160×1.5 is 30KB!

Palomino34, 2021-02-18:
We implemented 12BPP.

Note: Image only correct when 4 bytes (565-2pixels) to 3 bytes (444 - 2pixels).

Meaning X and Width of an object on image must be multiply of 2, if want object is displayed correctly.

closed 2021-02-18 by Palomino34

Palomino34, 2021-02-19:
Work perfectly now, no need x or width must be multiply 2.