ghi-electronics/TinyCLR-Libraries #831: NetworkStream does not handle zero-length return from Socket.Send

martincalsyn, 2021-01-16:
If Socket.Send can’t send all of the buffer you give it, it returns the number of bytes that it actually sent, and we mostly fixed NetworkStream.Write to handle that in preview3.  However, if Socket.Send can't send any part of the data (due to buffers in the underlying hardware being unavailable) then Socket.Send returns 0.  This isn't really an error (and Socket.Send should throw if there's an actual error, according to the docs for System.Net).  Instead, this means we should retry.

The fix that is needed here is that when Socket.Send returns 0, we should retry some number of times, with a small delay to allow the underlying buffers to clear out.

martincalsyn, 2021-01-16:
Pull request #832

Palomino34, 2021-02-17:
Done by PR. Thanks Martin

closed 2021-02-17 by Palomino34