ghi-electronics/TinyCLR-Libraries #833: ST7735Controller.DrawBuffer - Exception at higher SPI frequencies

taylorza, 2021-01-18:
Preview 3 has improved the situation, but I still get intermittent InvalidOperationException's raised on calls to ST7735Controller.DrawBuffer in the Graphics.OnFlushEvent handler. The issue seems more pronounced when debugging, but has been observed when the debugger is not attached as well. The issue, on occasion occurs within minutes of running the code and in other cases does not occur even after running for hours at a time. I am currently using an SPI frequency of 20Mhz, but I have seen the issue at 12Mhz and 15Mhz as well.

**Version**: TinyCLR 2.1.0 Preview 3
**Exception Details**: Exception thrown: 'System.InvalidOperationException' in GHIElectronics.TinyCLR.Devices.Spi.dll

taylorza, 2021-01-23:
Some additional information. I have had this issue occur at the 4Mhz as well. 

It appears that the issue occurs partway through the SPI transfer. I see a split screen update, where the upper region is the new frame that has transferred and the lower region, from the point where the exception was raised is the previous frame data. 

Since my frame updates are running as quickly as possible, I found that if I catch and ignore the exception the issue resolves within 2 frames, ie. the exception is raised on the next frame after the first exception but on the third frame things seem to have re-synced. I was considering re-initializing the SPI interface after the exception but I wanted to share this. 

Here is my code to flush the buffer

```
private static void Graphics_OnFlushEvent(Graphics sender, byte[] data, int x, int y, int width, int height, int originalWidth) {
    try {
        st7735.DrawBuffer(data, x, y, width, height);
    } catch(Exception ex) {
        Debug.WriteLine(ex.ToString());
    }
}
```

Here is a more complete stack trace:

```
#### Exception System.InvalidOperationException - CLR_E_INVALID_OPERATION (1) ####
    #### Message: 
    #### GHIElectronics.TinyCLR.Devices.Spi.Provider.SpiControllerApiWrapper::Write [IP: 0000] ####
    #### GHIElectronics.TinyCLR.Devices.Spi.SpiDevice::Write [IP: 0067] ####
    #### GHIElectronics.TinyCLR.Drivers.Sitronix.ST7735.ST7735Controller::DrawBuffer [IP: 001d] ####
    #### GHIElectronics.TinyCLR.Drivers.Sitronix.ST7735.ST7735Controller::DrawBuffer [IP: 001b] ####
    #### SC20100SIDE.DeviceLibrary::Graphics_OnFlushEvent [IP: 000f] ####
    #### System.Drawing.Graphics::Flush [IP: 0051] ####
```

MBNSoftware, 2021-01-31:
Chris, did you try to encapsulate your method in a lock ?
Not sure if it will help but it will prevent race conditions, at least.

taylorza, 2021-01-31:
> Chris, did you try to encapsulate your method in a lock ?
> Not sure if it will help but it will prevent race conditions, at least.

I had a similar suspicion and did try that, unfortunately it did not help.

Palomino34, 2021-02-12:
We just tested a simple project as below and could not reproduce at 24MHz.
```
static int SCREEN_WIDTH = 160;
static int SCREEN_HEIGHT = 128;
static ST7735Controller st7735;

static void Main()
{
   DrawOnN18();
}
static void DrawOnN18()
{
    var spi = SpiController.FromName(SC20100.SpiBus.Spi4);
    var gpio = GpioController.GetDefault();

    var customConfig = ST7735Controller.GetConnectionSettings(SpiChipSelectType.Gpio, GpioController.GetDefault().OpenPin(SC20100.GpioPin.PD10));

    customConfig.ClockFrequency = 24_000_000;

    st7735 = new ST7735Controller(
        spi.GetDevice(customConfig), // ChipSelect 
        gpio.OpenPin(SC20100.GpioPin.PC4), // Pin RS
        gpio.OpenPin(SC20100.GpioPin.PE15) // Pin RESET

    );

    var bl = gpio.OpenPin(SC20100.GpioPin.PA15); // back light

    bl.Write(GpioPinValue.High);
    bl.SetDriveMode(GpioPinDriveMode.Output);

    st7735.SetDataAccessControl(true, true, false, false); // rotate the screen
    st7735.SetDrawWindow(0, 0, SCREEN_WIDTH, SCREEN_HEIGHT);
    st7735.Enable();
    
    // Create flush event 
    Graphics.OnFlushEvent += Graphics_OnFlushEvent;

    var font = Resources.GetFont(Resources.FontResources.small);
    var cnt = 0;

    var colorW = new SolidBrush(Color.White);
    var colorR = new SolidBrush(Color.Red);
    var colorB = new SolidBrush(Color.Blue);

    var screen = Graphics.FromImage(new Bitmap(SCREEN_WIDTH, SCREEN_HEIGHT));

    while (true)
    {
        screen.Clear();

        screen.DrawString(cnt + " Hello Red! ", font, colorR, 5, 10);
        screen.DrawString(cnt + " Hello Blue! ", font, colorB, 5, 20);

        cnt++;
        screen.Flush(0, 0, SCREEN_WIDTH, SCREEN_HEIGHT);

        GC.Collect();
        GC.WaitForPendingFinalizers();

    }
}

private static void Graphics_OnFlushEvent(Graphics sender, byte[] data, int x, int y, int width, int height, int originalWidth) {
    st7735.DrawBuffer(data, x, y, width, height);

}
```

taylorza, 2021-02-12:
Dat,

Here is a screen shot of the issue and the code that I used to reproduce the issue with. Note, this can run for hours without causing an issue and other times the issue occurs after just a few minutes. One more thing, it also seems that as the system is under more load the more likely this is to happen, for example if I am using UsbHost with a keyboard attached the system slows down and the issue occurs more often.

[image]

```
using GHIElectronics.TinyCLR.Devices.Gpio;
using GHIElectronics.TinyCLR.Devices.Spi;
using GHIElectronics.TinyCLR.Drivers.Sitronix.ST7735;
using GHIElectronics.TinyCLR.Pins;
using System;
using System.Diagnostics;
using System.Drawing;
using System.Threading;

namespace SPIDisplayIssue
{
    class Program
    {
        static int SCREEN_WIDTH = 160;
        static int SCREEN_HEIGHT = 128;
        static ST7735Controller st7735;
        
        static void Main()
        {
            var spi = SpiController.FromName(SC20100.SpiBus.Spi4);
            var gpio = GpioController.GetDefault();

            SpiConnectionSettings customConfig = ST7735Controller.GetConnectionSettings(SpiChipSelectType.Gpio, GpioController.GetDefault().OpenPin(SC20100.GpioPin.PD10));

            customConfig.ClockFrequency = 20_000_000;

            st7735 = new ST7735Controller(
                spi.GetDevice(customConfig), // ChipSelect 
                gpio.OpenPin(SC20100.GpioPin.PC4), // Pin RS
                gpio.OpenPin(SC20100.GpioPin.PE15) // Pin RESET

            );

            var bl = gpio.OpenPin(SC20100.GpioPin.PA15); // back light

            bl.Write(GpioPinValue.High);
            bl.SetDriveMode(GpioPinDriveMode.Output);

            st7735.SetDataAccessControl(true, true, false, false); // rotate the screen
            st7735.SetDrawWindow(0, 0, SCREEN_WIDTH, SCREEN_HEIGHT);
            st7735.Enable();

            Graphics.OnFlushEvent += Graphics_OnFlushEvent;

            var whitePen = new Pen(Color.White);
            var blackPen = new Pen(Color.Black);

            var screen = Graphics.FromImage(new Bitmap(SCREEN_WIDTH, SCREEN_HEIGHT));

            var x = 80;
            var y = 64;
            var dx = 2;
            var dy = 2;
            var r = 5;
            var r2 = 2 * r;

            while (true)
            {
                screen.Clear();
                screen.DrawEllipse(whitePen, x - r, y - r, r2, r2);
                x += dx;
                y += dy;
                if (x <= r || x >= SCREEN_WIDTH - r) dx = -dx;
                if (y <= r || y >= SCREEN_HEIGHT - r) dy = -dy;
                screen.Flush();
            }
        }

        private static void Graphics_OnFlushEvent(Graphics sender, byte[] data, int x, int y, int width, int height, int originalWidth)
        {
            st7735.DrawBuffer(data, x, y, width, height);
        }
    }
}

```

Palomino34, 2021-02-12:
Sweet, we will check again.

Palomino34, 2021-02-12:
How long will the issue be happened after start the program? I mean after 1 min, 10 min...?

It is still working fine to me. I am using newest firmware which is a bit different what you have, it could be a reason.

closed 2021-02-12 by Palomino34

taylorza, 2021-02-12:
@Palomino34, the issue is intermittent which is why you are struggling to catch it. As mentioned earlier, it can run for hours before occurring and other time it can happen within 5 minutes. But, it always happens at some point.

Palomino34, 2021-02-12:
That is why I could not see. I thought it will be happened in shortly.

taylorza, 2021-02-12:
One thing, I do see is that if the system is busy (USB Host and CDC also enabled) it happens much more frequently.

I currently catch the exception and ignore it, but have a short beep every time it occurs so that I know it is happening. It happens more frequently in the primary application and since enabling both USB Host and CDC it happens even more frequently.

taylorza, 2021-02-13:
I ran some tests and captured the times of the exceptions, I restarted the application under the debugger after each exception. The exceptions occurred as follows:

54 min after start
8 min after start
12 min after start
7 min after start

taylorza, 2021-02-27:
@Palomino34, did you manage to reproduce this? I will run overnight to see if the issue still exists.

Palomino34, 2021-02-27:
Hi Chris,

Yes, the exception (overrun) is ignored if write only.