ghi-electronics/TinyCLR-Libraries #838: USB Host cause system slower.

Palomino34, 2021-01-20:
Not verified yet.

taylorza, 2021-01-23:
I have encountered this issue and have determined that the reason for the slowdown is due to the timer for CheckEvents firing at the default of every 10ms. 

As an example, for the Keyboard device I am able to increase the WorkerInterval to 50ms and even 100ms without impacting the user experience and get a noticeable improvement in overall performance of the rest of the system. Naturally this will not be a reasonable solution for other device types like a mouse etc. which would be noticeably impacted by the higher latency of the poll rate.

Palomino34, 2021-02-05:
How slow is it, 25%, 50% slower?

I am using N18, attach a thumbdrive or keyboard, even type few thing on keyboard, I don't see any different about the speed when drawing on screen.

gus-ghielec, 2021-02-05:
I think it is only like 10% and only on as very busy system.

taylorza, 2021-02-06:
I have not measured the % speed difference, but I can see it with "CPU" intensive work on the SC20100S Dev I can see the screen updates are visually slower and when I increase the WorkerInterval things get better. In my case rendering each frame requires a lot of behind the scenes processing so that would be amplifying the impact of the regular context switch I guess.

taylorza, 2021-02-12:
I got a moment to connect this up to the scope. It looks like the USB polling locks the system for 20ms, so every WorkerInterval period, so the default is that every 10ms the system is locked up for 20ms.
[image]

Here is the code I used to test

```
using GHIElectronics.TinyCLR.Devices.Gpio;
using GHIElectronics.TinyCLR.Devices.UsbHost;
using System.Threading;

namespace USB_KBD_Speed
{
    class Program
    {
        static void Main()
        {
            var gpio = GpioController.GetDefault();
            var usb = UsbHostController.GetDefault();
            usb.OnConnectionChangedEvent += Usb_OnConnectionChangedEvent;
            usb.Enable();

            var buzzer = gpio.OpenPin(GHIElectronics.TinyCLR.Pins.SC20100.GpioPin.PC6);
            buzzer.SetDriveMode(GpioPinDriveMode.Output);

            while (true)
            {
                buzzer.Write(GpioPinValue.High);
                Thread.Sleep(1);
                buzzer.Write(GpioPinValue.Low);
                Thread.Sleep(1);
            }
        }

        private static Keyboard keyboard;
        private static void Usb_OnConnectionChangedEvent(UsbHostController sender, DeviceConnectionEventArgs e)
        {
            if (e.DeviceStatus == DeviceConnectionStatus.Connected && e.Type == BaseDevice.DeviceType.Keyboard)
            {
                keyboard = new Keyboard(e.Id, e.InterfaceIndex);
            }
        }
    }
}

```

TGlev, 2021-02-16:
How would one increase the WorkerInterval time? I cant find this property anywhere.

Palomino34, 2021-02-16:
You can do keyboard.WorkerInterval = 50; // or 100.

TGlev, 2021-02-16:
So this is only a issue with keyboards/mice? I thought I could do the same for my USB drive, that one seems to lack the option.

Palomino34, 2021-02-16:
Any device need interval to response valid data.

TGlev, 2021-02-16:
> Any device need interval to response valid data.

Is there a WorkerInterval property for a usb flash drive? If so, where is it?

Palomino34, 2021-02-16:
It means (and in theory) thumb drive should be OK.

Palomino34, 2021-02-24:
Fixed.

closed 2021-02-24 by Palomino34