ghi-electronics/TinyCLR-Libraries #851: StreamReader.ReadLine() sometimes hang

bauland, 2021-02-02:
With 2.1.0 - Preview3, when I use:
```
            using (var fileStream = new FileStream(indexPath, FileMode.Open, FileAccess.Read, FileShare.Read))
            {
                using (var sr = new StreamReader(fileStream))
                {
                    while ((path = sr.ReadLine()) != null)
                    {
                         // Something to do with path...
                    }
                }
            }

```
Readline() function hangs sometimes (I can't find the cause).

Palomino34, 2021-02-16:
We need to know more detail, something such as what kind of the media, what size of file, and how often the issue happen...
A a small project that we can reproduce. 
Moved to v2.2.0 for now.

bauland, 2021-02-19:
I use sd card with FEZ Duino.

Here is code for a small project:
using System.Diagnostics;
using System.IO;
using System.Threading;
using GHIElectronics.TinyCLR.Devices.Storage;
using GHIElectronics.TinyCLR.IO;
using GHIElectronics.TinyCLR.Pins;

namespace testSdReadFileTxt
{
    class Program
    {
        static void Main()
        {
            var sdController = StorageController.FromName(STM32H7.StorageController.SdCard);
            var drive = FileSystem.Mount(sdController.Hdc);

            var path=@"A:\.Library\Index.txt";

            using (var fileStream = new FileStream(path, FileMode.Open, FileAccess.Read, FileShare.Read))
            {
                using (var stream = new StreamReader(fileStream))
                {
                    while (stream.Peek()>=0)
                    {
                        var str=stream.ReadLine();
                        Debug.WriteLine(str);
                    }
                }
            }

            FileSystem.Unmount(sdController.Hdc);

            Debug.WriteLine("OK"); // Sometimes never reached

            while (true)
            {
                Thread.Sleep(20);
            }
        }
    }
}

Text file contains:
Les vidéos vous permettent de faire passer votre message de façon convaincante. Quand vous cliquez sur Vidéo en ligne, vous pouvez coller le code incorporé de la vidéo que vous souhaitez ajouter. Vous pouvez également taper un mot-clé pour rechercher en ligne la vidéo qui convient le mieux à votre document.
Pour donner un aspect professionnel à votre document, Word offre des conceptions d’en-tête, de pied de page, de page de garde et de zone de texte qui se complètent mutuellement. Vous pouvez par exemple ajouter une page de garde, un en-tête et une barre latérale identiques. Cliquez sur Insérer et sélectionnez les éléments de votre choix dans les différentes galeries.
Les thèmes et les styles vous permettent également de structurer votre document. Quand vous cliquez sur Conception et sélectionnez un nouveau thème, les images, graphiques et SmartArt sont modifiés pour correspondre au nouveau thème choisi. Quand vous appliquez des styles, les titres changent pour refléter le nouveau thème.
Gagnez du temps dans Word grâce aux nouveaux boutons qui s'affichent quand vous en avez besoin. Si vous souhaitez modifier la façon dont une image s’ajuste à votre document, cliquez sur celle-ci pour qu’un bouton d’options de disposition apparaisse en regard de celle-ci. Quand vous travaillez sur un tableau, cliquez à l’emplacement où vous souhaitez ajouter une ligne ou une colonne, puis cliquez sur le signe plus.
La lecture est également simplifiée grâce au nouveau mode Lecture. Vous pouvez réduire certaines parties du document et vous concentrer sur le texte désiré. Si vous devez stopper la lecture avant d’atteindre la fin de votre document, Word garde en mémoire l’endroit où vous avez arrêté la lecture, même sur un autre appareil.

Here is output window: (Note that OK is never reached and last line is not read)
```
   Assembly: GHIElectronics.TinyCLR.Native (2.1.0.0)  Attaching deployed file.

   Assembly: GHIElectronics.TinyCLR.Devices.Storage (2.1.0.0)  Attaching deployed file.

   Assembly: testSdReadFileTxt (1.0.0.0)  Resolving.

The debugging target runtime is loading the application assemblies and starting execution.
Ready.

'GHIElectronics.TinyCLR.VisualStudio.ProjectSystem.dll' (Managé) : 'D:\Nicolas\source\repos\testSdReadFileTxt\testSdReadFileTxt\bin\Debug\pe\..\GHIElectronics.TinyCLR.IO.dll' chargé, chargement des symboles ignoré. Le module est optimisé et l'option du débogueur 'Uniquement mon code' est activée.
'GHIElectronics.TinyCLR.VisualStudio.ProjectSystem.dll' (Managé) : 'D:\Nicolas\source\repos\testSdReadFileTxt\testSdReadFileTxt\bin\Debug\pe\..\GHIElectronics.TinyCLR.Native.dll' chargé, chargement des symboles ignoré. Le module est optimisé et l'option du débogueur 'Uniquement mon code' est activée.
'GHIElectronics.TinyCLR.VisualStudio.ProjectSystem.dll' (Managé) : 'D:\Nicolas\source\repos\testSdReadFileTxt\testSdReadFileTxt\bin\Debug\pe\..\GHIElectronics.TinyCLR.Devices.Storage.dll' chargé, chargement des symboles ignoré. Le module est optimisé et l'option du débogueur 'Uniquement mon code' est activée.
'GHIElectronics.TinyCLR.VisualStudio.ProjectSystem.dll' (Managé) : 'D:\Nicolas\source\repos\testSdReadFileTxt\testSdReadFileTxt\bin\Debug\pe\..\testSdReadFileTxt.exe' chargé, symboles chargés.
Le thread '<Aucun nom>' (0x2) s'est arrêté avec le code 0 (0x0).
Les vidéos vous permettent de faire passer votre message de façon convaincante. Quand vous cliquez sur Vidéo en ligne, vous pouvez coller le code incorporé de la vidéo que vous souhaitez ajouter. Vous pouvez également taper un mot-clé pour rechercher en ligne la vidéo qui convient le mieux à votre document.
Pour donner un aspect professionnel à votre document, Word offre des conceptions d’en-tête, de pied de page, de page de garde et de zone de texte qui se complètent mutuellement. Vous pouvez par exemple ajouter une page de garde, un en-tête et une barre latérale identiques. Cliquez sur Insérer et sélectionnez les éléments de votre choix dans les différentes galeries.
Les thèmes et les styles vous permettent également de structurer votre document. Quand vous cliquez sur Conception et sélectionnez un nouveau thème, les images, graphiques et SmartArt sont modifiés pour correspondre au nouveau thème choisi. Quand vous appliquez des styles, les titres changent pour refléter le nouveau thème.
Gagnez du temps dans Word grâce aux nouveaux boutons qui s'affichent quand vous en avez besoin. Si vous souhaitez modifier la façon dont une image s’ajuste à votre document, cliquez sur celle-ci pour qu’un bouton d’options de disposition apparaisse en regard de celle-ci. Quand vous travaillez sur un tableau, cliquez à l’emplacement où vous souhaitez ajouter une ligne ou une colonne, puis cliquez sur le signe plus.
```

bauland, 2021-02-19:
I've tried to change content of text file. Try to change sd card.

Palomino34, 2021-02-19:
Thanks, can you tell me what version are you using?

bauland, 2021-02-19:
Yes it's 2.1.0 - Preview3

Palomino34, 2021-02-19:
Thank you. Its is fixed.

commit 659f024f (2021-02-20): Fixed #851 - File stream hangs sometime
commit f07af273 (2021-02-20): Merge pull request #892 from ghi-electronics/fixed_851_fs_hang

Fixed #851 - File stream hangs sometime
closed 2021-02-19 by Palomino34