ghi-electronics/TinyCLR-Libraries #878: Image/Bitmap inconsistent implementation

taylorza, 2021-02-13:
`System.Drawing.Image` implements `SetPixel`, which is unnecessarily reimplemented in `System.Drawing.Bitmap` and inadvertently hiding the base implementation. Bitmap also implements `GetPixel `which could be moved up to Image to be consistent with where `SetPixel `is implemented.

Note: In the full .NET Framework, `GetPixel` and `SetPixel` are not implemented in Image, only in `Bitmap` but if for design reasons `SetPixel` it is required in Image then it should be consistent and have `GetPixel` as well.

Palomino34, 2021-02-17:
Done

closed 2021-02-17 by Palomino34

taylorza, 2021-02-20:
@dat, why override get and set pixel in Bitmap? The implementation is exactly the same as the inherited implementation. Not a big deal, but I was curious if there is a performance or some other benefit relating to the TinyCLR internals that would be useful to know.

commit 82d718da (2021-02-16): Fixed #878: Get and Set Pixel in Image/Bitmap
commit ee1ad53e (2021-02-16): Merge pull request #884 from ghi-electronics/image_bitmap_inconsistent_implementation

Fixed #878: Get and Set Pixel in Image/Bitmap