ghi-electronics/TinyCLR-Libraries #880: CDC on linux

Palomino34, 2021-02-15:
https://forums.ghielectronics.com/t/cdc-on-linux/23571

taylorza, 2021-02-19:
I have looked into this and I have created a CDC implementation that works on both Windows and Linux.

Here is the endpoint configuration I used

```
var readEndpoint = this.ReserveNewEndpoint();
var writeEndpoint = this.ReserveNewEndpoint();
var interruptEndpoint = this.ReserveNewEndpoint();

Configuration.Endpoint[] endpoints =
{
    new Configuration.Endpoint((byte)interruptEndpoint, Configuration.Endpoint.ATTRIB_Write | Configuration.Endpoint.ATTRIB_Interrupt) { wMaxPacketSize = 16, bInterval = 250},
    new Configuration.Endpoint((byte)writeEndpoint, Configuration.Endpoint.ATTRIB_Write | Configuration.Endpoint.ATTRIB_Bulk) { wMaxPacketSize = 64 },
    new Configuration.Endpoint((byte)readEndpoint, Configuration.Endpoint.ATTRIB_Read | Configuration.Endpoint.ATTRIB_Bulk) { wMaxPacketSize = 64 },
}
```

**However** there seems to be another issue, the bmAttributes are not 100% compliant. For example, when you define an Out endpoint the high bit of the endpoint address is being set, but the bmAttributes high bit is also set. Bits 7..2 are reserved and should be set to 0. Windows is very forgiving, but Linux does not like this. 

To verify this, I added the following code after setting the `endpoints` which adjusts the endpoint address appropriately and resets the reserved bits of the `bmAttributes`

```
foreach(var ep in endpoints)
{
    if ((ep.bmAttributes & 0x80) != 0) ep.bEndpointAddress |= 0x80;
    ep.bmAttributes &= 3;
}
```

I did not want to provide this as a pull request, because of the hack I needed to do with the attributes. I think this is a design flaw in the interface using the bmAttributes to specify the endpoint direction since that is not compliant with the specs. BUT I am not a USB expert at all so I might be wrong.

Palomino34, 2021-02-21:
Thanks, we corrected it.

closed 2021-02-21 by Palomino34