ghi-electronics/TinyCLR-Libraries #914: DrawString exception in tight loop

microcompiler, 2021-03-02:
Using the following sample code DrawString() is throwing System.NotSupportedException.   The strange thing about this exception is that when you run the same code sometimes it takes a minute to throw other times just a few seconds.   Also if you have extended memory enable it takes longer but after several minutes it will throw.  Delaying the loop to 200 milliseconds will delay how long before the exception is thrown however if you wait long enough it will throw.  I did try using a timer with the same results.   

I have attached the full sample project with font and a screen shot of the exception.  [drawrain.zip](https://github.com/ghi-electronics/TinyCLR-Libraries/files/6066292/drawrain.zip)

Using v2.1.0-preview3 packages and the FEZ portal Rev C (same results on Dev Board).

```csharp
class Program
{
    static readonly Font robotoFont = Resources.GetFont(Resources.FontResources.RobotoFont);

    static readonly Random rand = new Random();

    static void Main()
    {
        var backlight = GpioController.GetDefault().OpenPin(SC20260.GpioPin.PA15);
        backlight.SetDriveMode(GpioPinDriveMode.Output);
        backlight.Write(GpioPinValue.High);

        var controller = DisplayController.GetDefault();
        controller.SetConfiguration(new ParallelDisplayControllerSettings
        {
            Width = 480,
            Height = 272,
            DataFormat = DisplayDataFormat.Rgb565,
            Orientation = DisplayOrientation.Degrees0,
            PixelClockRate = 10000000,
            PixelPolarity = false,
            DataEnablePolarity = false,
            DataEnableIsFixed = false,
            HorizontalFrontPorch = 2,
            HorizontalBackPorch = 2,
            HorizontalSyncPulseWidth = 41,
            HorizontalSyncPolarity = false,
            VerticalFrontPorch = 2,
            VerticalBackPorch = 2,
            VerticalSyncPulseWidth = 10,
            VerticalSyncPolarity = false,
        });

        controller.Enable();

        var width = controller.ActiveConfiguration.Width;
        var height = controller.ActiveConfiguration.Height;
        var display = Graphics.FromHdc(controller.Hdc);

        while (true)
        {
            display.DrawString("@", robotoFont, new SolidBrush(Color.White), rand.Next(width), rand.Next(height));
            display.DrawString("!", robotoFont, new SolidBrush(Color.Blue), rand.Next(width), rand.Next(height));
            display.DrawString("$", robotoFont, new SolidBrush(Color.Green), rand.Next(width), rand.Next(height));
            display.DrawString("%", robotoFont, new SolidBrush(Color.Red), rand.Next(width), rand.Next(height));
            display.DrawString("&", robotoFont, new SolidBrush(Color.Yellow), rand.Next(width), rand.Next(height));
            display.DrawString("#", robotoFont, new SolidBrush(Color.Purple), rand.Next(width), rand.Next(height));
            display.DrawString("*", robotoFont, new SolidBrush(Color.Black), rand.Next(width), rand.Next(height));
            display.DrawString("+", robotoFont, new SolidBrush(Color.Gray), rand.Next(width), rand.Next(height));
            display.DrawString("?", robotoFont, new SolidBrush(Color.Teal), rand.Next(width), rand.Next(height));
            display.Flush();

            Thread.Sleep(5); 
            
            // This loop will run for less then a minute then throw System.NotSupportedException
            // The time will vary anywhere from a few seconds to a minute
        }
    }
}
```

Palomino34, 2021-03-02:
You need to move "new SolidBrush(Color.White)", new SolidBrush(Color.Blue)..... out of the loop

Or try to add

```
while (true) {
   display.DrawString.....
    .....

  GC.Collect();
  GC.WaitForPendingFinalizers();
}
```

closed 2021-03-02 by Palomino34

microcompiler, 2021-03-02:
Totally makes sense.  Rookie move on my part. I should have caught that. :)

Closed #914.