ghi-electronics/TinyCLR-Libraries #917: Ethernet PHY options

gus-ghielec, 2021-03-03:
The TI PHY we use now is same one used for the last 10 years. Works great but not good on low power. Look into a secondary option for Ethernet PHY.

For example, the one ST uses on their  [demo board](https://www.st.com/resource/en/user_manual/dm00385071-evaluation-boards-with-stm32h743xi-and-stm32h753xi-mcus-stmicroelectronics.pdf) and on [discovery board](https://www.st.com/content/ccc/resource/technical/layouts_and_diagrams/schematic_pack/group1/79/99/be/65/3c/76/40/d6/MB1364-H743ZI-C01_Schematic/files/MB1364-H743ZI-C01_Schematic.pdf/jcr:content/translations/en.MB1364-H743ZI-C01_Schematic.pdf) which is [LAN8742A](https://www.microchip.com/wwwproducts/en/LAN8742A)

skeller2015, 2021-03-08:
Maybe consider Single Pair Ethernet (SPE)?  Not sure if there are PHY's that can do both 100/1000Base-T1 and 10BASE-T1L.

https://e2e.ti.com/blogs_/b/analogwire/posts/how-10base-t1lsingle-pair-ethernet-phys-help-bring-more-data-farther-in-long-distance-applications

gus-ghielec, 2021-05-11:
An option DP83825I