ghi-electronics/TinyCLR-Libraries #939: Brush and pen

gus-ghielec, 2021-03-16:
Brush vs pen vs color. Only one is necessary

AndreMarschalek, 2021-03-17:
Brush and Pen are different things, Brush used to fill area, Pen to draw font or Border or whatever and Color require both?

gus-ghielec, 2021-03-17:
I know what but why not use a pen for everything? Why do we need gradient fill? This is an embedded system and things should be efficient and lean. Why create an object for color to draw something? 

Maybe it's us correct. But I think we should investigate.

AndreMarschalek, 2021-03-17:
i saw gradient fill only in demos 
basically its true, you need background color, foreground color and thickness of whatever, that's it from my point of view
and use gradient fill (and other) resources for something more useful