ghi-electronics/TinyCLR-Libraries #940: StreamReader.Peek() hangs without error

microcompiler, 2021-03-17:
When reading using StreamReader Peak() attached to a NetworkStream the system hangs without any error. To recover you have to reset the board. I tested this on both Preview3 and Preview 4.  The only differences was the firmware and the preview packages.  Using the most common methods to read a line from StreamReader is have varying results on the preview 4 release. 

Full project: [reader-issue.zip](https://github.com/ghi-electronics/TinyCLR-Libraries/files/6153588/reader-issue.zip)
Tested On: SCM20260D Dev Board

This method works without any issue on preview 3 and 4:
```csharp    
    // This works on both preview 3 and 4
    string line;
    do
    {
        line = reader.ReadLine();
        Debug.WriteLine(line);
    } while (!reader.EndOfStream);
```
These methods would not work on preview 4 and would hang the system without any error:
```csharp    
    // This works on preview 3 only and hangs at "ReadLine()" on preview 4              
    string line;
    while ((line = reader.ReadLine()) != null)
    {
        Debug.WriteLine(line);
    }

    // This works on preview 3 only and hangs at "Peek()" on preview 4  
    while (reader.Peek() != -1)
    {
        var line = reader.ReadLine();
        Debug.WriteLine(line);
    }
```

Full Code:

```csharp
class Program
  {
      static void Main()
      {
          // Initialize SC2026D development board ethernet
          InitializeEthernet();
  
          var server = new HttpServer();
          server.Start();
      }
  
      private static void InitializeEthernet()
      {
          var networkController = NetworkController.FromName(
              SC20260.NetworkController.EthernetEmac);
  
          var networkInterfaceSetting = new EthernetNetworkInterfaceSettings
          {
              MacAddress = new byte[] { 0x00, 0x8D, 0xA4, 0x49, 0xCD, 0xBD },
              DhcpEnable = true,
              DynamicDnsEnable = true
          };
  
          networkController.SetInterfaceSettings(networkInterfaceSetting);
          networkController.NetworkAddressChanged += NetworkAddressChanged;
          networkController.SetAsDefaultController();
  
          networkController.Enable();
      }
  
      private static void NetworkAddressChanged(
          NetworkController sender,
          NetworkAddressChangedEventArgs e)
      {
          var ipProperties = sender.GetIPProperties();
          var address = ipProperties.Address.GetAddressBytes();
  
          if (address != null && address[0] != 0 && address.Length > 0)
          {
              Debug.WriteLine($"Lauch web brower on: http://{ipProperties.Address}");
          }
      }
  }

public class HttpServer
  {
      private Thread _thread;
      private Socket _listener;

      private bool _active = false;
      private readonly ManualResetEvent _acceptEvent = new ManualResetEvent(false);
      private readonly ManualResetEvent _startedEvent = new ManualResetEvent(false);

      public void Start()
      {
          // Don't return until thread that calls Accept is ready to listen
          _startedEvent.Reset();

          // create the socket listener
          _listener = new Socket(AddressFamily.InterNetwork, SocketType.Stream, ProtocolType.Tcp);
          _listener.SetSocketOption(SocketOptionLevel.Socket, SocketOptionName.ReuseAddress, true);
          _listener.SetSocketOption(SocketOptionLevel.Socket, SocketOptionName.KeepAlive, true);

          // bind the listening socket to the port
          IPEndPoint endPoint = new IPEndPoint(IPAddress.Any, 80);
          _listener.Bind(endPoint);

          // start listening
          _listener.Listen(5);

          _thread = new Thread(() =>
          {
              _active = true;
              AcceptConnections();
          });
          _thread.Priority = ThreadPriority.AboveNormal;
          _thread.Start();

          // Waits for thread that calls Accept() to start
          _startedEvent.WaitOne();

          Debug.WriteLine($"Started socket listener");
      }

      public void Stop()
      {
          _active = false;

          // Signal the accept thread to continue
          _acceptEvent.Set();

          // Wait for thread to exit 
          _thread.Join();
          _thread = null;

          _listener.Close();
          _listener = null;

          Debug.WriteLine("Stopped socket listener");
      }

      private void AcceptConnections()
      {
          // Set the started event to signaled state
          _startedEvent.Set();

          while (_active)
          {
              // Set the accept event to nonsignaled state
              _acceptEvent.Reset();

              Debug.WriteLine("Waiting for a connection...");
              using (var remoteSocket = _listener.Accept())
              {
                  // Set the accept event to signaled state
                  _acceptEvent.Set();

                  // Send response to client
                  Response(remoteSocket);

                  // Close connection
                  remoteSocket.Close();
              }

              // Wait until a connection is made before continuing
              _acceptEvent.WaitOne();
          }

          Debug.WriteLine("Exited AcceptConnection()");
      }

      private void Response(Socket socket)
      {
          using (var network = new NetworkStream(socket))
          {                
              using (var reader = new StreamReader(network))
              {
                  // This works on both preview 3 and 4
                  string line;
                  do
                  {
                      line = reader.ReadLine();
                      Debug.WriteLine(line);
                  } while (!reader.EndOfStream);

                  // This works on preview 3 only and hangs at "ReadLine()" on preview 4              
                  //string line;
                  //while ((line = reader.ReadLine()) != null)
                  //{
                  //    Debug.WriteLine(line);
                  //}

                  // This works on preview 3 only and hangs at "Peek()" on preview 4  
                  //while (reader.Peek() != -1)
                  //{
                  //    var line = reader.ReadLine();
                  //    Debug.WriteLine(line);
                  //}

                  string response = "HTTP/1.1 200 OK\r\nContent-Type: text/html; charset=UTF-8\r\n\r\n" +
                                       "<doctype !html><html><head><title>Hello, world!</title>" +
                                       "<style>body { background-color: #111 } h1 { font-size:2cm; text-align: center; color: white;}</style></head>" +
                                       "<body><h1>" + DateTime.Now.Ticks.ToString() + "</h1></body></html>";

                  var bytes = Encoding.UTF8.GetBytes(response);
                  network.Write(bytes, 0, bytes.Length);
              }
          }
      }
  }
```

microcompiler, 2021-05-01:
Any chance this issue might make it into RC2.   As I look at this issue closer what I thought was a viable work around doesn't look to be.  I'm not sure how to track down the exact issue but even when I use the following method (that I though previously worked):

```CSHARP
string line;
do
{
    line = reader.ReadLine();
    Debug.WriteLine(line);
} while (!reader.EndOfStream);
```

It feels like it's blocking at inconsistence points in time.  The one way I can get it to fail every time is by using postman to POST a binary string like "0123456789".  It will "hang"  sometimes at the beginning of the string, sometimes on the very last charter.  Also,  using postman if you terminate the session it will finish the loop.  When I use the term "hang"  I now think it's some type of strange blocking going on. I'm struggling with other work arounds  to try fix this behavior.   I would really appreciate any thoughts you might have.   I'm planning on releasing NuGet packages base on the final 2.1 release but with this issue (and no work around) it's throwing a monkey wrench in that plan.

microcompiler, 2021-05-02:
I think I might have found the issue.   Should line 156 in StreamReader.cs set a -1 instead of a 1.  I'm pretty sure this issue has to with reading the last byte of the stream. If the last bytes do not containing a \n\r it's getting stuck in the while loop on line 103.
```CSHARP
154 if (readCount == 0)
155 {
156   readCount = -1;
157 }
```

Palomino34, 2021-05-03:
Set timeout. Starting preview4 it is become infinite as default that why you see hang.

microcompiler, 2021-05-04:
Thanks for your help!  This explains a lot.  As it was working previously without throwing an IOException. I never thought to look for the timeout properties.  This behavior is much closer to the full dot net stack from a blocking standpoint.  It's still not really doing what I want so I'm going to approach it completely differently not using StreamReader.

Palomino34, 2021-05-04:
Inherit from StreamReader, set timeout to 50ms, try cache the timeout exception, it should be same as preview3 or earlier.

microcompiler, 2021-05-05:
Thanks!  This is very helpful.

Palomino34, 2021-05-07:
Any luck with this issue? Share to us how you did if you can, so we can improve our end, please. :))

microcompiler, 2021-05-08:
I played around with capturing the exception for a while by crating a custom HttpStreamReader class (and a few other things) but no matter what I tried I could not pick up the very last character of the stream (I can see it in the network stream).  The exception is trapped but the ReadLine() doesn't contain the last charter (unless CRLF are the last).  I could also make it work by coping the network stream into a memory stream and then using StreamReader on the memory stream.     I'm guessing this is me not handling something properly.     However, In the end I got it working by not using StreamReader at all and processing the bytes myself (primarily to keep it efficient). 

From my perspective I have a working solution but I couldn't get StreamReader to do what I was looking for.  
   
```csharp
public class HttpStreamReader : StreamReader
    {
        public HttpStreamReader(Stream stream) 
            : base(stream)
        {
            stream.ReadTimeout = 50;
        }

        public override string ReadLine()
        {
            try
            {
                return base.ReadLine();
            }
            catch (Exception)
            {
                // Eat exception
                return null;
            }
        }
    }
```

microcompiler, 2021-12-03:
@Palomino34  I wanted to let you know that I think we can safely close this issue.  This turned out to be more of my specific use case.   I found some time to really dig into this and it turned out to be a timing issue processing the HTTP request.  The connection is established but I was then trying to read the header data before it was fully transmitted.   I added a "WaitForData" method that checks the socket.Available waiting for the http header to be completely transmitted (or timeout).   This way the do/while loop includes the final CRLF and loop completes.   

I have tested this with 200,000+ requests and it's worked flawlessly.  Here is the actual implementation in case you are interested.   I really appreciate your help with this issue.       

https://github.com/bytewizer/microserver/blob/develop/src/Bytewizer.TinyCLR.Http/Internal/HttpMiddleware.cs
https://github.com/bytewizer/microserver/blob/develop/src/Bytewizer.TinyCLR.Http/Internal/HttpMessage.cs

closed 2021-12-03 by microcompiler