ghi-electronics/TinyCLR-Libraries #945: Application.Lock brings in GHIElectronics.TinyCLR.Networking

taylorza, 2021-03-19:
The need to lock an application requires GHIElectronics.TinyCLR.Update which depends on GHIElectronics.TinyCLR.Networking, this increases the deployment size by 12,564 bytes even if networking is not required. In my case I am only locking the device to gain access to the CDC and WinUSB.

Palomino34, 2021-03-31:
Network and Pin nugets are removed.

taylorza, 2021-04-02:
@Palomino34 - Ideally we should pull the Application class out of the GHIElectronics.TinyCLR.Update assembly. If, as in my case, the only need is to Lock the application to access WinUSB, this still pulls in GHIElectronics.TinyCLR.IO which is +/- 20KB, which is needed for the IFU support.

gus-ghielec, 2021-04-02:
Reworked the API for better interface and moved to native. See #971

closed 2021-04-02 by gus-ghielec