ghi-electronics/TinyCLR-Libraries #948: TinyCLR Config should recognize multiple firmware versions, or generate errors on mismatch

martincalsyn, 2021-03-22:
The TinyCLR Config program should validate the firmware version that it is operating against when setting options like 'extended memory' and 'extended flash', etc.

It should either generate an error if it is being used to set options on an incompatible memory map, OR, should recognize the firmware version and adjust accordingly.

At the least, it should not make adjustments to the wrong location and the user should not be given the illusion of success when, in fact, the setting did not 'take'.

gus-ghielec, 2021-03-23:
I believe config has always been the same and so any TinyCLR version will work on any 2.x release.

Are you sure there is a problem?

TGlev, 2021-03-23:
Try opening the Device Configuration form using TinyCLR config preview 4 while having a device with firmware preview 3 attached. It crashes because it fails to read the device name, I think.

martincalsyn, 2021-03-23:
Yes, there is definitely a problem between versions and even between previews.  Symptoms range from 'it crashes' to 'it fails silently'.  Flashing works - it's just the Device Config as @TGlev  says.

AndreMarschalek, 2021-03-28:
time to open source TinyCLR Config :-)

Palomino34, 2021-04-05:
For now will simply state the version mismatch but will not handle. This is usually a concern with preview releases and not production.

closed 2021-04-05 by Palomino34