ghi-electronics/TinyCLR-Libraries #968: DHCP with reconnect

gus-ghielec, 2021-04-01:
Does TinyCLR automatically re-lease am IP on reconnect or user needs to reset the stack?

Does the stack terminates all connections when a connection is lost?

Related post https://forums.ghielectronics.com/t/apipa-autoip/23003/47

Palomino34, 2021-04-01:
Just quick test, could not reproduce. Result below:
```
Disconnected. Wait for connecting.... 20
Disconnected. Wait for connecting.... 30
Mac: 70:b3:d5:fa:c0:0f
ip address :192.168.86.37
subnetmask :255.255.255.0
gate way :192.168.86.1
dns[0] :75.75.75.75
dns[1] :75.75.75.76
Phy staus True
Connected.... 40
Connected.... 50
Connected.... 60
Connected.... 70
Mac: 70:b3:d5:fa:c0:0f
ip address :0.0.0.0
subnetmask :0.0.0.0
gate way :0.0.0.0
dns[0] :0.0.0.0
dns[1] :0.0.0.0
Phy staus False
Disconnected. Wait for connecting.... 80
Disconnected. Wait for connecting.... 90
Disconnected. Wait for connecting.... 100
Mac: 70:b3:d5:fa:c0:0f
ip address :192.168.86.37
subnetmask :255.255.255.0
gate way :192.168.86.1
dns[0] :75.75.75.75
dns[1] :75.75.75.76
Phy staus True
Connected.... 110
Connected.... 120
Connected.... 130

```

Palomino34, 2021-04-01:
We mean user no need to reset the stack.

closed 2021-04-01 by Palomino34