ghi-electronics/TinyCLR-Libraries #1076: SC13: Add backupmemory

Palomino34, 2021-09-01:
Use bk registers for this memory

Palomino34, 2021-09-08:
Done

closed 2021-09-08 by Palomino34

axa88, 2021-09-12:
To confirm this is enabling RTC memory?
Will this get a target version?

Palomino34, 2021-09-13:
Yes, but only 76 bytes for user only on SC13. It is backup registers but we can read/write single byte as well.

axa88, 2021-09-17:
@Palomino34

I'm not familiar with "backup registers", my question is, in any way besides `total size`, does this deviate from the RTC memory description from:
https://docs.ghielectronics.com/software/tinyclr/tutorials/real-time-clock.html#battery-backed-memory

Also is there a target version?
Thanks

Palomino34, 2021-09-17:
No different. And you even don't need to know they are "backup registers", if you don't care.