ghi-electronics/TinyCLR-Libraries #1084: Add JsonIgnore attribute

TGlev, 2021-09-29:
Is it possible to get a json ignore attribute?

This way we can set properties in a class to not be serialized.

martincalsyn, 2022-07-20:
Looks like this is not possible. You can create and set attributes on TinyClr classes, props, methods, fields, etc., but the runtime does not support enumerating them.

Attributes in mscorlib and the TinyClr libraries all seem to be either compile-time hints, or things that the runtime uses internally (e.g., MethodImplAttribute). The MemberInfo, MethodInfo, etc., classes do not include the .GetCustomAttributes reflection call that we need.

Palomino34, 2022-07-20:
Won't fixed as [martincalsyn](https://github.com/martincalsyn) recommended.

closed 2022-07-20 by Palomino34

TGlev, 2022-07-20:
@Palomino34 Would it be possibvle to add the GetCustomAttributes reflection method?