ghi-electronics/TinyCLR-Libraries #1161: Change JSON order to order in class

TGlev, 2022-05-23:
As discussed with Martin, I'm creating the issue for this.

martincalsyn, 2022-07-20:
This looks to not be possible right now.  I have a code change that will preserve order between deserialization and re-serialization, and this also means that props and fields are emitted in the right order, but props are emitted together and fields are emitted together. The reason for this is that the TinyClr type system provides GetMethods and GetFields but not GetMembers (which would return fields and props together).  So, I have to retrieve one (Props as methods) and then the other (fields).

However, interspersing fields and props is messy C# anyway and most people are probably using one or the other exclusively, or are clumping the fields and the props together anyway.

So, the current change is a big improvement. It provides deser/ser round tripping, and it orders things correctly within props and fields - just not across props and fields.

TGlev, 2024-04-22:
Martin's fix has solved the issue for me completely.

closed 2024-04-22 by TGlev