ghi-electronics/TinyCLR-Libraries #1330: Incorrect Double.ToString formatting up to an thrown Exceptions.

cdwork, 2023-12-14:
[Original post "Incorrect Double.ToString formatting up to an thrown Exceptions"](https://forums.ghielectronics.com/t/incorrect-double-tostring-formatting-up-to-an-thrown-exceptions/24796)

Double.ToString using formatting throws an exception.
For example:
```
(0.1).ToString("F3");                     // Work correct
(0.6-0.5).ToString("F3");                 // throw exception
(0.099999999999999978).ToString("F3");    // throw exception
```
because of
[`var result = FormatNative(value, formatCh, precision);`](https://github.com/ghi-electronics/TinyCLR-Libraries/blob/7a88c7eb71a8de4a73889ca51c34194e05a35281/mscorlib/Number.cs#L278C26-L278C67)
in [System.Number.Format(object value, bool isInteger, string format, NumberFormatInfo info)](https://github.com/ghi-electronics/TinyCLR-Libraries/blob/7a88c7eb71a8de4a73889ca51c34194e05a35281/mscorlib/Number.cs#L275)
will return a string, firstly, that does not match the value, and secondly, with “non-text” characters.
```
value = 0.099999999999999978
formatCh = "F"
precision = 3
result = "0.011000000000000000000000000000000000000000000000000 \b$] 0p0�a�0�aD"
```
different values reproduce exceptions with different specified formatting precision.

Palomino34, 2023-12-14:
link on forum: https://forums.ghielectronics.com/t/incorrect-double-tostring-formatting-up-to-an-thrown-exceptions/24796

Palomino34, 2023-12-15:
seem duplicate but need verify: https://github.com/ghi-electronics/TinyCLR-Libraries/issues/629

closed 2024-03-14 by Palomino34