ghi-electronics/TinyCLR-Libraries #1343: Sending too many TCP messages hangs entire system

TGlev, 2024-04-22:
As discussed over mail, sending too much TCP will hang entire system until TCP connection is dropped.
Below is a sample project to reproduce the issue:
https://github.com/TGlev/TCPStackGhiTest

TGlev, 2024-04-23:
Can be the same. I will try the workaround.

TGlev, 2024-04-23:
This prevents the system from hanging completely. But when opening a putty window, only a first NMEA sentence is received and then nothing more because the poll function will always return false.

Palomino34, 2024-04-25:
If you don't need delay then you can try 
```
public virtual void OpenConnection() {
while (true)
{
    try
    {
        //Wait for a connection
        Socket ClientSocket = Socket.Accept();

        ClientSocket.SetSocketOption(SocketOptionLevel.Tcp, SocketOptionName.NoDelay, true);

        //If we make it this far, a connection was been established!
        IsConnected = true;

....

```

We tried and it works fine.

Another way, sender sends message to receiver and wait for the response from that receiver, before sending a new message.

closed 2024-04-25 by Palomino34