ghi-electronics/TinyCLR-Libraries #1360: Generate TCA file even when deploying to a device in Visual Studio

skeller2015, 2024-05-16:
Generate the .tca file even when Transport = USB and deploying to the device when in Visual Studio.

This eliminates the additional steps required to use TinyCLR Config application to generate the tca file.

Palomino34, 2024-05-17:
It would be nice if have this, but most of the time users don't need this, it will slow down or effect the deployment

Just select "generate tca and raw...." and user will have tca file.

skeller2015, 2024-05-17:
I understand it slows the deployment but a simple check box to have it as an option would be helpful.   This would be similar to other embedded development environments that generate a usable output file on each compile.

Palomino34, 2024-06-17:
In stead of checking on the check box, user can switch them in the combobox.

closed 2024-06-17 by Palomino34