ghi-electronics/TinyCLR-Libraries #1361: Add File Version to TCA file properties in Windows Explorer

skeller2015, 2024-05-16:
When viewing the .tca files properties in Windows Explorer -> Properties -> Details add File Version and have it match the Assembly version.

Palomino34, 2024-05-17:
what is "Windows Explorer -> Properties -> Details add File Version"? 
Can you take a screen shot?

skeller2015, 2024-05-17:
Right-click on a file in Explorer and select Properties, then go to Details Tab.  This is just showing a standard EXE file.  Currently, a TCA file has no File Version info associated with it.  Having the correct Version info throughout the process makes it easier to identify the proper files that need to be used, especially during the development stage.

[image]

Palomino34, 2024-05-17:
I thought you are saying that we can select TCA file's property and can see the version. Now I understand what you mean.

We don't plan to change TCA format,  it is same all the time (till now).  No mater what Config version, the format are same.

skeller2015, 2024-05-17:
I am not looking for the format of the TCA file to change.  It would be helpful for the version file property to be added to the file.  As you can see from most of my requests in the last few days there is no consistency in the TCA file version.  The value entered in TinyCLR Config shows up when you do a verify, but it won't/can't match the  Assembly.GetName().Version info in code so there is no easy way in code to determine if the TCA file is newer than the current running version.  In TinyCLR Config you can't add the full version numbers, like 1.1.8903.34567.    The current procedures to generate files for In-Field updates are not friction free.  Especially during the development and testing phase.   If the TCA file is generated in Visual Studio then the Version file attribute should be able to be added.  The current file version should also be available as it gets set somewhere during the compile phase since it is based on the *current date and time.  (*When using the format 1.1.*._ in Visual Studio Assembly Version.)

Palomino34, 2024-06-17:
I see the benefit of this but it can't not apply to tca file format.

closed 2024-06-17 by Palomino34